uorocketry/rocket-code-2020 #53: When testing, the states in input.txt and output are not the same

sebastiengrd, 2021-04-07:
It looks like the states are shifted by one timestamp in the output of testing.

For example, in 0.87.uorocketlog
925248900780, 3, ...
925282233113, 3, ...
925315562020, 5, ...
925348910613, 6, ...
925382243446, 6, ...

But in input.txt
925248900780, ... , 3, ...
925282233113, ... , 3, ...
925315562020, ... , 3, ...
925348910613, ... , 5, ...
925382243446, ... , 6, ...

Note: the log format are different for those two files

ajayyy, 2021-04-07:
Yep, I noticed this, and am pretty sure it is due to https://github.com/uorocketry/rocket-code-2020/issues/52

sebastiengrd, 2021-04-07:
I'm not soo sure... The initialization event occurred at the beginning in the test case. But we also see the shift later in the log files (when the transition are based on data of events, e.g. liftoff, that are logged)

ajayyy, 2021-04-07:
Another potential reason could be related to header logic ignoring the first line of the CSV.

ajayyy, 2021-09-29:
Since #52 is fixed, this issue may be fixed now