uorocketry/rocket-code-2020 #67: Switch servo control to use a more complex data transmission system

ajayyy, 2021-05-01:
Servo control needs to send more than just boolean data, which isn't possible in the current binary format. Look into using a different format.

Possible options:

http://bsonspec.org/

https://github.com/protocolbuffers/protobuf

kovalromank, 2021-05-01:
Can you give some examples of the type of data you want to send? Right now it accepts boolean data because each valve is either open or closed. Do we want the servo control to control more than just the valves?

ajayyy, 2021-06-07:
https://jpa.kapsi.fi/nanopb/

ajayyy, 2021-06-07:
Alternative to protobuf that seems faster to encode/decode: https://capnproto.org/index.html

It seems protobuf can be slow

Another alternative is https://avro.apache.org/

Both of these use schemas files like protobuf, capnproto generates code likke protobuf, Apache Avro does not.

jonapap, 2022-05-09:
Superseded by #191