uorocketry/rocket-code-2020 #177: Use pre-existing code from other libraries to reduce complexity

SeoFernando25, 2022-03-05:
eg: 

`stringSplit(text, delimeter)` can be replaced by `boost::split(tet, line, boost::is_any_of(delimiter));`

https://github.com/uorocketry/rocket-code-2020/blob/f0fe2147216709eddbad1525e38c33b2b8d3c635/src/helpers/Helper.cpp#L8

`XXXX processXXXX(data)` can be replaced by `boost::lexical_cast<uint64_t>(data);`

https://github.com/uorocketry/rocket-code-2020/blob/f0fe2147216709eddbad1525e38c33b2b8d3c635/src/helpers/Helper.h#L20

`SharedArray<T>` can be made safer by replacing it by `std::shared_ptr<std::vector<T>>`
https://github.com/uorocketry/rocket-code-2020/blob/f0fe2147216709eddbad1525e38c33b2b8d3c635/src/helpers/Helper.h#L31

`EventQueue` can be replaced by `boost::lockfree::queue<eventType>`

https://github.com/uorocketry/rocket-code-2020/blob/f0fe2147216709eddbad1525e38c33b2b8d3c635/src/IO/EventQueue.h#L7

`GpioState` and `GpioData` can be merged (by only having `digitalMap` and `pwmMap` variables)

https://github.com/uorocketry/rocket-code-2020/blob/f0fe2147216709eddbad1525e38c33b2b8d3c635/src/data/GpioData.h#L5-L15

 Please don't do this, use `std::min` instead

https://github.com/uorocketry/rocket-code-2020/blob/f0fe2147216709eddbad1525e38c33b2b8d3c635/src/helpers/Types.h#L5-L6

https://github.com/uorocketry/rocket-code-2020/blob/b13b8ae260e07490b476c0996f48999f03ba06bf/src/stateMachine/InterfacingStateMachine.cpp#L18

jonapap, 2022-04-15:
So, I have a few comments regarding the items that are left:
1. We have to define "safer" here. Yes, `vector` is safer in the sense that it uses dynamic allocation, will grow as needed, and doesn't require the use of any pointers. However, in the specific case where `SharedArray` is used (`ArduinoEncoder`), we need to have a way to request a specific block of memory and have access to raw pointers. While `vector` could be used, it would be weird, error-prone, and would anyway not offer any advantages.
2. No, let's keep using locks. From what I understand, `boost::lockfree` in our scenario would actually be **less** performant than using a mutex.  In short, `boost::lockfree` uses atomic CPU operations instead of a lock to synchronize access. Those operations are pretty slow. Therefore, those structures should only be used in certain specific scenarios. 
3. Could you elaborate on how those two `GPIO` structs could be merged?

If you have questions regarding the first two, please ask. I know those can be tricky concepts to understand.

SeoFernando25, 2022-06-30:
You're totally correct about points 1 and 2.
I think we have discussed about the GPIO structs in a meeting that its better to keep them like that for clarity sake. 

I think that was about it, thanks! :smile:

closed 2022-06-30 by SeoFernando25