uorocketry/rocket-code-2020 #191: Explore using MAVLink for communication

jonapap, 2022-04-15:
Right now, we are using a somewhat hacky way to inject RSSI packets into our serial stream. However, it looks like the RFD900+ can inject RSSI info if it detects that we are using the MAVLink protocol. If it detects MAVLink heartbeat packets, it will start injecting that RSSI information using MAVLink.

Also, it seems MAVLink could replace Protobuf. It has the similar idea of defining messages, along with a code generator. It can even be used for both TCP and Radio, making the codebase easier to deal with.

ajayyy, 2022-05-09:
https://mavlink.io/en/services/camera.html could be useful for remote video

jonapap, 2022-05-09:
Hum, could be worth looking into. However, wouldn't we want a separate frequency for video? I don't think we should mix critical info, like GPS, with video.